BillyBlaze/OctoPrint-TouchUI #501: OctoPrint Touch UI menu buttons disappear completely

jhbeskow, 2021-06-30:
Hello,

I have an issue that started occurring across all browsers today.  Any that are using Touch UI can no longer display the menu buttons at the top.  I do not know for sure if this is the issue or not, but I am receiving a message in developer tools stating "Refused to apply style from 'http://*****.local/static/webassets/packed_plugins.less?d4c89092' because its MIME type ('application/octet-stream') is not a supported stylesheet MIME type, and strict MIME checking is enabled."  I get that whenever I disable TouchUI, as well, so maybe not.  Everything else seems to work fine.  I have attempted to uninstall and re-install Touch UI, but that did not seem to help.  Do you have any thoughts on how to fix it outside of rebuilding the OctoPi device?

Thank you!

Testmangh, 2021-07-02:
Mine disappeared too =(

BillyBlaze, 2021-07-04:
The issue has been fixed on the maintenance branch and will be released a.s.a.p

closing this issue since its a duplicate of #494

closed 2021-07-04 by BillyBlaze