BillyBlaze/OctoPrint-TouchUI #502: xserver setup by https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart prevents using screen rotation.

scyto, 2021-07-01:
When I use Octoprint OS image and touch ui using the https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart scripts to install X it results in an X implementation where it is impossible to rotate the screen either with `xrandr` as per https://www.raspberrypi.org/documentation/configuration/display_rotation.md or with the gui tool arandr as expected.

When using desktop version of raspbian this all works perfectly.  It seems that the script https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/helpers/install-steps/4-xorg-setup is likely not doing enough config or https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/helpers/install-steps/3-apt-get-install isn't installing enough.

Unfortunately I don't know enough about X (until today all my 5+ years of pi experience has been SSH only) to suggest a fix.  But happy to help where I can.

scyto, 2021-07-01:
Ok to summarize how to get rotation working for HDMI displays.

1. the /f99-fbden.conf has to not exist
2. screen has to be rotated using one of these two approaches:
- use `sudo arandr` (can be installed using apt) this has advantage of being a UI and writing settings to /usr/share/dispsetup.sh
- use xrandr as per raspi orgs docs
3. during startup either /usr/share/dispsetup.sh has to be called (after X has started) and before app starts OR use xrandr to rotate manually

Once all of these steps have been done, the touch matrix can be re-orientated to match by adding one of these 3 options too `/usr/share/X11/xorg.conf.d/40-libinput.conf `

90° = Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"
180° = Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
270° = Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1"  

for example, on my system in this section in that file

```
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"
EndSection
```

scyto, 2021-07-01:
:-)
[image]

BillyBlaze, 2021-07-05:
Hi, I have stopped giving instructions about rotating displays since they change alot and can be different with other touchscreens. So the best way is to check the manufacturing instructions.

You can also add a file `calibration.sh` inside the ` TouchUI-autostart` folder. This file will be executed before launching the browser. (see [code here](https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/chromium.xinit#L84))

As for changing the font-size just navigate to `Settings > TouchUI > Enable customization` and select your prefered size in the dropdown. If you want to tweak it more then you need to add a local LESS file and change the variable to your liking from there. More on the latter part is over here: https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Customize:-Use-your-own-file

closed 2021-07-05 by BillyBlaze

scyto, 2021-07-06:
@BillyBlaze i understand your position.
For reference the method described above is generic for all HDMI based monitors that use HDMI cable and that use USB to connect the screen as a touchpad - there is no calibration for example.  This is only about rotation.  Either way i solved it and this issue is here for posterity for others.  

Thanks for the tip on the script in the TouchUI-autostart, that's perfect.

I saw the LESS instructions before, read them and left clueless about where a LESS file goes, what its format is etc.  It assumes i know what a LESS file is (googling generic LESS instructions helped but i still didn't understand where i was supposed to make those edits.  Not sure if you can be a little more explicit in the linked instructions for noob idiots like me. :-)

Nice project, thanks for replying, appreciate it.