BillyBlaze/OctoPrint-TouchUI #505: Navbar lost after update

InoxDragoon, 2021-07-06:
when using Touch UI on my phone and via octoeverywhere on my laptop, i lost the navigation bar, making the plugin almost useless. This happened today, 7/6/21.
[image]

thedavefulton, 2021-07-06:
I'm seeing exactly the same behavior on both mobile devices and the 7" touchscreen attached to the pi

BillyBlaze, 2021-07-08:
What OctoPrint version is running? This was just fixed in #494 for OctoPrint 1.6.1

thedavefulton, 2021-07-08:
I'm on `1.6.1`

For my part everything seems to be working fine now. I haven't had any issues past the initial problem right after updating the plugin.

InoxDragoon, 2021-07-08:
Octoprint 1.6.1. It had me reboot after the update, and that's when things went wonky. However, after a subsequent reboot (also had to reinstall some MIA plug-ins after the update) all is well now. I should note that I've only been using octoprint for a few weeks now, so pardon me if I've missed an unwritten rule about rebooting after a reboot.... Lol

Thanks!

closed 2021-07-30 by github-actions[bot]