BillyBlaze/OctoPrint-TouchUI #507: TouchUI Website is not displayed on TFT

kami83, 2021-07-15:
Hi,

since some time my chromium browser is not displaying the small TFT TouchUI website on my LCD3,5. It is only showing the Big Octopi-Website?

Can someone help. 

BG kami

BillyBlaze, 2021-07-15:
Hi,

Check the following (if the answer is yes move to the next one):

1. is TouchUI still installed?
2. is TouchUI still active? (e.g. not disabled)
3. is TouchUI working on other devices?

kami83, 2021-07-15:
Hi, TouchUI is installed. On the LCD i cannot find a button to activate it. On the other devices like my laptop i can activate TouchUI and it is working.

BillyBlaze, 2021-07-15:
hmm thats really weird.

1. have you modified the bootloader? (if yes; try reverting it to the original state)
2. could you disable all plugins and only activate TouchUI, restart TFT and see if it comes back

Fox2525, 2021-07-16:
Hi, I have got the exact same problem. I have got a 7 inch touch display, that normally worked fine. After some updates it only opens the normal Octoprint Desktop Website. I also can't toggle back on the screen. With my Laptop and handy I can toggle back and forth and touch ui is working fine there. Haven't made any modifications on the bootloader. I will try to disable soms plugins and see if the probldm is still there.

kami83, 2021-07-16:
Hi, yes there is something wrong in the 0.3.18 version. I reinstalled an older Backup and everything works fine. Only after updating to Octopi 1.6.1 and 0.3.18 i get on my small touchdisplay the same Webview as on my normal browser PC.

How can i switch it. I think TouchUI is not recognizing the small size of the Screen. On my PC Browser i can active and deactived TouchUI normal. Thanks a lot.

Cu kami

BillyBlaze, 2021-07-16:
is the plugin CustomizerUI perhaps installed?

kami83, 2021-07-16:
No, this is my plugin list:
 "Firmware Updater",
  "DisplayZ",
  "BetterHeaterTimeout",
  "OctoPod Plugin",
  "Custom Control Editor",
  "TouchUI",
  "Fan Speed Control",
  "Bed Visualizer",
  "Complicated Plugin",
  "SlicerSettingsParser",
  "Filament Sensors Revolutions",
  "Dashboard",
  "Marlin EEPROM Editor",
  "PrintTimeGenius Plugin",
  "Filament Manager",
  "Octolapse"

kami83, 2021-07-21:
Hi the problem comes from Filament Manager Plugin. If i deactive it, everything works fine.

Thx. 

BG kami

closed 2021-07-21 by kami83