BillyBlaze/OctoPrint-TouchUI #517: Fails to Install

CBaker007, 2021-11-08:
I am running the latest version of OctoPi.   When I boot, I get the tty login on the touchscreen. 

I follow the direction here:
https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser

1) I install the plugin on OctoPi web interface.
2) I have a 3.5" but using other drivers so skip
3) Clone git.
4) Then I run the command to install but it fails.

pi@octopi:~ $ sudo ~/TouchUI-autostart/helpers/install
[sudo] password for pi:
 ââ¦âââââ¦ â¦ââââ¦ â¦â¦ â¦â¦
  â â ââ ââ  â ââ£â ââ
  â© âââ
Install helper v0.0.4

Running apt-get update (this will take a while)
Do NOT switch off the Pi or close this console until done!
 - failed

What is wrong and how do I fix or debug?

sirdezzy, 2021-11-13:
Try;

sudo apt update --allow-releaseinfo-change
sudo apt upgrade --fix-missing

Then run the install script again.

Chaukk, 2021-11-17:
> Try;
> 
> sudo apt update --allow-releaseinfo-change sudo apt upgrade --fix-missing
> 
> Then run the install script again.

This worked for me thank you.

rcardenaz, 2021-11-19:
Thanks guys for the help! I could not get it running and missed that "Issues" was where to go if you had any problems. I reinstalled OctoPi like 3 different times trying to get it to work. 
I will say that I did try @Chaukk's and it said something like this does not support arguments. I then tried sirdezzy's and it worked fine.

Chaukk, 2021-11-20:
> Thanks guys for the help! I could not get it running and missed that "Issues" was where to go if you had any problems. I reinstalled OctoPi like 3 different times trying to get it to work.
> I will [...]

I’m not sure what you mean by “Chaukk’s” I was simply replying to sirdezzy.

closed 2021-12-13 by github-actions[bot]