BillyBlaze/OctoPrint-TouchUI #521: Install Helper failed

ClueControl, 2021-12-30:
Hi, I get the following when running install helper with debug:

Install helper v0.0.4
DEBUG mode enabled
Failed to stop touchui.service: Unit touchui.service not loaded.

Running apt-get update (this will take a while)
Do NOT switch off the Pi or close this console until done!
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists... Done
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed i                             
ts 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be ap                             
plied. See apt-secure(8) manpage for details.

wfendler, 2022-01-14:
I'm getting the same response. Using a raspberry pi 4.

```bash
pi@octopi:~ $ sudo ~/TouchUI-autostart/helpers/install
[sudo] password for pi: 
 ╔╦╗╔═╗╦ ╦╔═╗╦ ╦╦ ╦╦ 
  ║ ║ ║║ ║║  ╠═╣║ ║║ 
  ╩ ╚═╝╚═╝╚═╝╩ ╩╚═╝╩ 
Install helper v0.0.4

Running apt-get update (this will take a while)
Do NOT switch off the Pi or close this console until done!
 - failed
```

```bash
pi@octopi:~ $ sudo apt-get update && sudo apt-get upgrade --yes
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists... Done      
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```

@ClueControl - have you had any luck?

wfendler, 2022-01-14:
Looks like this suggestion works: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/517#issuecomment-968066840

Run these two commands
```
sudo apt update --allow-releaseinfo-change
sudo apt upgrade --fix-missing
```

Then re-run
```
sudo ~/TouchUI-autostart/helpers/install
```

closed 2022-01-22 by github-actions[bot]