BillyBlaze/OctoPrint-TouchUI #532: Temp Graph in TouchUI 0.3.18 obscured after upgrade to Octoprint 1.8.0

MarkForrest, 2022-05-26:
Upgraded to Octoprint 1.8.0, using Raspberry Pi 4 plus official touchscreen in portrait orientation, temperature graph is shifted lower on screen so it can't be seen.
Tried updating Raspberry Pi OS.
Verified by doing fresh install of OctoPi 0.18 with Octoprint 1.8.1
Also forced TouchUI mode on laptop running recent Chromium and Firefox.
On Firefox the graph isn't obscured but is pushed down the page

Only fixed by downgrading Octoprint back to 1.7.3 and temp graph was then corrected.

vexofp, 2022-05-29:
I think the culprit here may be the new "wrench" menu button added in Octoprint 1.8.0 (the one that currently allows you to toggle "Show state marks"). It is implemented as a new `div` above the temperature graph. It looks like TouchUI is expecting the temperature graph to be `div:first-child` in that context, when it adjusts it to take up most of the screen. Now that first child is the button's row, instead, so we end up with the graph pushed down off the bottom.

LooLaBoo, 2022-06-14:
Yes I was thinking the same thing that the culprit was the "new wrench". I'm interested to see your results MarkForrest. I'd have a try myself but i've not enough coding know how to make those kinds of edits. Good luck.

vexofp, 2022-06-14:
Fixing the styling for Octoprint 1.8.x is fairly easy; [here is a branch that does so](https://github.com/BillyBlaze/OctoPrint-TouchUI/compare/master...vexofp:532-temp-graph-1.8.x). (Just changes the selector to the second child, instead of the first. I also made the new wrench button overlay on top of the graph, instead of wasting precious space above it.)

The catch is that this will break the layout for older Octoprint versions before 1.8. I am not sure how to gracefully handle both versions at the same time, unfortunately. @BillyBlaze Do we need to worry about such backwards compatibility, or is upping the required Octoprint version acceptable? (Or does anyone have a good idea how to handle both layouts?)

closed 2022-06-21 by github-actions[bot]