BillyBlaze/OctoPrint-TouchUI #534: heat graph

varek3d, 2022-06-23:
Hi, love this plugin, however, for some reason, I can no longer see the heat chart that shows how much it's heating.  I tried to open this on a regular chrome browser and it showed but rather squished and lower down on the screen.  This doesn't interfere with printing at all, but I thought you may want to know.

yet-another-average-joe, 2022-06-25:
Same here. Haven't been using the printer for a couple months, and needed it today. Just updated OctoPrint, and the problem occured.

OctoPi 0.18.0
OctoPrint 1.8.1

No problems when TouchUI is turned off (7" HDMI touch LCD) ; browser = Chromium, window manager = OpenBox. (custom TouchUI installation, no desktop environment)

varek3d, 2022-06-27:
Ahh, the update.. Interesting, those pesky updates!

yet-another-average-joe, 2022-06-27:
1.7.3 : no problems

1.8.0 introduced a new graph feature : markers, and the problem with TouchUI (just verified)

Workaround : rolled back to OctoPrint 1.7.3

varek3d, 2022-06-27:
that's just as good as not using the touch ui...wth!

davidgamero, 2022-06-27:
Temp graph seems to have a bug i can see the whole graph is pinned to the very bottom on safari mobile
[image]
[image]

TisDone, 2022-07-12:
Same thing is happening to me ... any chance for an update on whether this is resolvable?  
Running on a Raspberry Pi 4 w/ 4 GB Ram
Running Raspberry OS Buster
Octoprint 1.8.1
TouchUI (0.3.18)

varek3d, 2022-07-12:
really?  So the programer can't help?

vexofp, 2022-07-17:
See #532 , which was an earlier issue for this same problem. It ended up auto-locked due to inactivity. (In fact, every issue since #512 has met the same fate, and this one will likely soon follow.) I suspect @BillyBlaze may have stepped away from the project, given the lack of any activity this year.

I made a [branch](https://github.com/BillyBlaze/OctoPrint-TouchUI/compare/master...vexofp:532-temp-graph-1.8.x) which fixes this problem (but breaks compatibility with Octoprint <1.8). Works well for me, but YMMV.

davidgamero, 2022-07-17:
@vexofp thanks for the link I'll give it a try! Did you ever get a solution for backwards compatibility? Im on latest, but just curious

closed 2022-07-25 by github-actions[bot]